RapierXbox/ShellyElevate #1: Could you provide some examples? :)

DcFamas, 2024-07-21:
Hey there,
I want to buy and test the Shelly Wall Display with ShellyElevate. Unfortunately I don't know how the handling is right now. Does it work really good? Could you maybe provide some pictures of the finished Shelly? Thanks and best regards :)

RapierXbox, 2024-07-26:
Yeah, sorry that I read your issue just now... I can take some pictures tomorrow.

RapierXbox, 2024-07-26:
Here is it in Home Assistant: 
[image]

NoUsername10, 2024-07-28:
Hi, great work.
Could you add pictures of the display running HA also if possible?

RapierXbox, 2024-08-04:
https://github.com/user-attachments/assets/dfbf606d-deba-43f9-ac13-08e8eea46190 Here you go. You can hear the relay clicking.

closed 2024-08-04 by RapierXbox

DcFamas, 2024-08-05:
Great, thank you! :)
Will buy one and test and give some feedback.