RapierXbox/ShellyElevate #4: Kiosk-Mode or supressing adress bar?

Fritz-la-chatte, 2024-08-09:
Hi, I watched your short video with the Home Assitant interface with great enthusiasm.

Soon, thanks to your instructions, I would like to try putting the HA on the Shelly Wall Display. 
I noticed in your video that the address bar is still visible in the browser.
https://github.com/user-attachments/assets/dfbf606d-deba-43f9-ac13-08e8eea46190

Is it possible to suppress this with a fully kiosk mode in Chrome, to gain more working space and better overview on the display?

Another approach might be to combine this FullyKiosk app, if the manufacturer allows it.
https://www.home-assistant.io/integrations/fully_kiosk/

Just an idea, but first of all, I want to thank you for all your effort!

RapierXbox, 2024-08-13:
Hi, I noticed that too. When you scroll down, the search bar disappears. I was planning to use WebView in a separate application, but the old preinstalled version of WebView doesn't work well with Home Assistant. Updating WebView was tricky for me and often led to a boot loop. However, I'm working on a guide to help with this and will be publishing it soon.

Fritz-la-chatte, 2024-08-15:
That would certainly be great, as it would surely have a positive impact on the user experience.

Thank you for further optimizing this already great implementation!

KaHa6uc, 2024-08-27:
Hello, Shelly Wall Display lead dev here. Just a few notes:

First, great work, @RapierXbox 

Next, any browser that you install will use the webview that is currently installed. Stock webview cannot show the HA pages in a fully usable manner, no matter what browser you use.

Finally, there is a new Wall Display version coming with new hardware and the webview will be updated to a more recent version that will be able to show HA normally. With this new version HA will be supported internally.

Again, great work!

KaHa6uc, 2024-08-29:
@Fritz-la-chatte
This is really not the place for this, but I will nevertheless reply you, because you seem to be confused. Maybe I wasn't clear, and I will try to fix this.

Neither the Thermostat functionality, nor HomeAssistant integration were on the table when we first designed the Wall Display. We did find out there were issues with the temperature reading correctness when we received the first golden samples from the factory but initially those were overcome with an internal temperature offset. Only after we started implementing the Thermostat we found out that those readings were unreliable for this function. Hence we started offering free BLU H&T devices for every Wall Display purchased from the earliest batches. More info, [here](https://www.facebook.com/groups/ShellyIoTCommunitySupport/posts/7734726613293302). Is this appealing enough for you?

As for the webview, we never intended to actually use it. We're using a proprietary Android build from what is called "a solution company" and only designed and wrote the front-end application. What we had was enough until recently, when we decided that HomeAssistant integration is a must. That's when we found out that the webview is outdated as hell and HA would not work, almost at all. Although using a webview in a system process is strongly discouraged for security reasons ([here](https://android.googlesource.com/platform/frameworks/base/+/2dfb79a/core/java/android/webkit/WebViewFactory.java#96)), we are currently preparing an OTA update that will update the implementation, and after that there will be another update in which HomeAssistant will be integrated in this webview in the application, so no, your guess is not correct - native HA integration will be available for the first Wall Display version. This large update will not be needed for the new hardware version, which will come with a more recent webview implementation and native HA integration.

You know what's really disappointing? Putting up invectives without being informed. I'm sorry we have lost you as a client. Maybe we should have popped up a message on the screen saying you're eligible for a free BLU H&T? Oh, sorry, I forgot about `./adb shell pm disable cloud.shelly.stargate`...

Now that [I hope] I have made myself clear, let's keep this issue clean and refrain from any more off-topic comments.

Again, praises for the great work of RapierXbox!

RapierXbox, 2024-09-02:
Hi everyone,

Thanks again for your appreciation of my project. I’m reaching out to see if anyone with more knowledge might be willing to help me test the new version. I’ve figured out how to install a new system WebView, but the method I’m using is quite risky and could potentially result in a boot loop. I’m considering trying it out, but my family won't be thrilled if the lights don’t work for the next week since I don’t have a spare to test on, haha.

Also, it’s great to see someone from the Shelly team here!

RapierXbox, 2024-09-02:
> @KaHa6uc That's basically what RapierXbox has already said and is now trying to correct.
> 
> I actually also bought the wall display as a thermostat. In the meantime, people are suggesting that you shouldn't rely on the internal sensor and that you should integrate an external H&T. (very appealing...) Now, apparently, if I understand right, the money is to be taken out of the customer's pocket because the product is generally disappointing or unsuitable for many users and to invest money once again in a new hardware revision. An interesting approach. I'm sure that someone from the HA community, not least RapierXbox, will also be able to solve this problem. If my guess is correct, I don't understand why Shelly doesn't release an update with a full webview and support the community. Shamed be he who thinks evil of it.
> 
> For me, this product is likely to be my 1st & last from Shelly under these circumstances, as I could have bought a Chinese touchscreen with relay for around €35. On the hardware side, it would have been just as good, if not better. A real disappointment.

Also added official thermostat implementation in  the readme for you!

Fritz-la-chatte, 2024-09-09:
Hello RapierXbox,

I don't know how much I should tell you thank you for this effort.
After realizing that the internal sensor is a bit “fickle” and I don't see the point of coupling an external sensor, which I find unattractive, I switched all the classic GIRA thermostats to anthracite Homematic wall thermostats instead of the wall display.
Since then, I have been hoping for a worthwhile alternative use, whereby I then intend to sacrifice a “stupid” wall lightswitch and until then look forward to your fantastic solution of the HA dashboard with switching function.

So I can only repeat how great I think your work is!

Many thanks and best regards from Austria.

Fritz-la-chatte, 2024-10-17:
Hi RapierXbox, 
with all the joy of many Wall Disply users that at least a moderately functioning Home Assistant integration including a new WebView has now found its way into the official Shelly beta firmware, I wondered whether this might not have opened the door for you to disable the address bar or to operate a FullyKiosk mode?
Shelly is avoiding to hide a disturbing bar at the bottom of the screen. In their forum was written, that they will keep their bar because it should be a "core feature". Even if the majority don't like it. 
https://community.shelly.cloudtopic/2401-home-assistant-in-/wall-display/?do=findComment&comment=12583

Also, the HA dashboard is still rather slow while the workload is kept simple.
I would be pleased if you would continue to pursue your approach, as I believe that many users, like me, are no longer interested in the Shelly environment and would prefer pure HA control.
Best regards

RapierXbox, 2024-11-06:
Hi everyone,

I was considering abandoning this project, especially after noticing that Shelly had implemented a similar feature, which I assumed most people would prefer. Additionally, I haven’t had much time to dedicate to it recently. However, knowing that many of you are still interested in seeing this project move forward has motivated me to continue.

I will first check if the latest update includes a new version of WebView. If it does, I will resume development by removing the access bar, adding a swipe-to-toggle light feature, and introducing more improvements.

Thank you for your support!

RapierXbox, 2024-11-06:
`After realizing that the internal sensor is a bit “fickle” and I don't see the point of coupling an external sensor, which I find unattractive,` I am maby working on replacing the internal sensor for a higher quality one. No promises though!

RapierXbox, 2024-11-29:
5 Shelly displays orderd. Expect some updates soon!

closed 2025-02-16 by RapierXbox

Fritz-la-chatte, 2025-02-27:
Just wanted to say, that your work is great!
Kind regards from Austria.

RapierXbox, 2025-02-27:
Thanks from Germany! Don't forget to create an issue if you have any kind of feature request.