RapierXbox/ShellyElevate #6: Is there a 12V version? Mine starts with 12V, not 5V

Morveus, 2024-09-03:
Hi,

First of all, thank you for your work!

I tried tonight to solder wires to the TX/RX pads, tripped, and ripped off the conductive layer on the TX1 pad; I tried recovering by scraping off a bit of the PCB to expose the outer conducting part of the pad, soldered again, and after trying to power on a regulated 5V power supply my Wall Display booted on loop, showing the Shelly logo, then turning off, the power supply showing consumption going from a few watts to zero, then the device started up again.

I thought it was short-circuited, put back the original power supply and it booted back up again, working fine (no serial activity however).

So I measured the voltage between those two pins
[image]

And got 12V.

When I turn my power supply to 12V, the device booted and worked very well (still no luck with the serial port, probably due to the dead pad)

Does it mean there are 12V versions? Or am I missing something?

RapierXbox, 2024-09-04:
That's new to me. Could you send me a picture of your display so I can compare it with mine? I'm curious if there might be a different hardware revision. The only revision I'm aware of involves the antenna connectors, but I can't figure out why the wall display would need 12V.

RapierXbox, 2024-09-04:
Were you able to perform the hack even after the RX and TX pads were ripped off? A little trick I use is to work with magnet wire and tape it down—though I had the same issue the first time I tried it.

Morveus, 2024-09-05:
Thank you for your reply! Sorry I wasn't at home, I'll post a picture and measurements this weekend.

Morveus, 2024-09-06:
Here's my power supply sending 12V to the pin identified as 5V in your README.md
[image]

And pictures of the board, if that can help:
[image]

[image]

[image]

I've switched to the NSPanel Pro as it was faster to get developper privileges, and have given the Shelly to a friend, but hopefully this might help people who have the same revision (if I've not overlooked something?) and wonder why the device bootloops on 5V ;)

Andoramb, 2024-11-04:
Yeah... I've got 12V as well 🙂 
And the same board.
I cannot get serial to work properly, it only returns garbage characters... 
Tried with 3 different converters and all the possible combinations I can think of (baudrate, tx/rx and so on)
it should be simple, but something is strange here 🙂

RapierXbox, 2024-11-06:
For me, even after purchasing a fourth Shelly display specifically for this project, it still works seamlessly with my original 5V setup. Did you use the UART settings I provided in the documentation? If you have access to an oscilloscope or similar equipment, you could try measuring the UART voltage to see if you maby need a level shifter.

Andoramb, 2024-11-12:
Yeah, I've tried everything, UART voltage was around 1.7V.
I've  ended up doing the MTKClient dump.
Just needed a proper linux machine and it went fine 🙂 
Thanks for the stuff here 🙂

closed 2025-02-15 by RapierXbox