RapierXbox/ShellyElevate #8: No Temp or Humidity

Andoramb, 2025-02-21:
This is great, but I can't seem to get any metrics from the device.
After an update to V2 and fresh reboot, I cannot get anything on `:8080`
(but this was probably the case before the update as well)

Even `netstat -tulpn` shows that only port 5555 is used.

So I would like your opinion on 2 questions here 😁 
1. Which app is responsible for providing data?
2. Should I upgrade the whole Shelly FW on the device?
2.1. But will I loose root then?

I really like the documentation update here btw 👌🏻
I just feel like the updates should be managed somehow 🤔

RapierXbox, 2025-02-22:
Hi there,

The app that provides the data is a background service that is automatically started by the ShellyElevate app. So, you wouldn’t technically need to install the launcher; it’s just for the user experience.

When updating to Shelly firmware, you can't really lose root as the device is rooted by Shelly themselves, but you could loose adb access. However, as long as you can see the Home Assistant interface in the app, it should be fine (or if you installed the Home Assistant update that is integrated into Shelly).

Some questions about your app:

1. Does it start?
2. Could you start a logcat on the tag "ShellyElevateV2" and then trigger the boot completed event by running:
```sh
./adb shell am broadcast -a android.intent.action.BOOT_COMPLETED
```
and check if you see the server start message?
I am looking into adding a button to pull the latest update and install it, but the Shelly firmware with integrated Home Assistant is a requirement because it contains a new version of WebView that is needed for displaying Home Assistant.

Andoramb, 2025-02-25:
So, let's see 🙂 
I have sorted out the fw upgrade from Shelly, got my adb as well 👍 

I have updated the apk to the very latest and tried again, same results:
1. I can input my local HA IP, but nothing really happens after I press `Back` (I was looking for a confirmation button, say `Save`)
2. No HttpServer is running 🙂

I think I kinda lost the purpose of the app now... I was looking for a background service, that would provide the data, but should it open HA in a browser or something?

I'm asking this, because I've installed the HA companion app onto the Shelly, so now it acts as  a standalone HA instance 🙂

Great update to the API endpoint page btw, I really look forward using it.

I have also created the sensors with a twist:
```
sensor: 
  - platform: rest
    name: Shelly Walldisplay Temperature
    resource: !secret shellyWDTemp
    value_template: "{{ (value_json.temperature | float(default=0)-8)  | round(1)}}"
    method: GET
    unit_of_measurement: "°C"
    scan_interval: 120
```
Added `unit_of_measurement' and altered a bit the 'value_template' to show 'default 0'
Not sure, if rounding is needed, but we'll see once it get's to work 🤘🏻

RapierXbox, 2025-02-26:
OK, that's kind of weird. First of all, the app is both the Home Assistant Companion and a background server in one. It looks like your app doesn't start automatically—maybe. Does it start automatically, or do you need to launch it manually from the launcher?

Also, I'm probably going to push a release in a few hours (maybe 2–4) so you can start the HTTP server from the settings. The back button saves everything.

Did you make sure it's the correct IP? It should be in a format without "http" or a port—just the raw IP.

When installing the app, did you make sure to use the -g flag? Without it, the app doesn't have permission to start on boot.

Andoramb, 2025-02-26:
Thanks for the update.
OK, so I've found one issue: because I've changed my default port from `8123` to something else, so I thought the port was also necessary... Once changed back to default, it opened the HA interface 🙂
(got this through logcat: `WebView : loadUrl=http://192.168.1.100:8123`)

But then still, no other ports were used. Checked I've it with nmap, dumpsys and logcat.

RapierXbox, 2025-02-26:
No problem! Happy to help.
Note to myself: I am going to need to make it so you also enter the port.
Did you try updating to the new release with the `-g -r` flags? With that version, you can start the HTTP server in the GUI.

Also, does your app start automatically, or do you need to start it yourself?

Andoramb, 2025-02-28:
Alright, I managed to get it to work now 🙂
Updated to the latest apk, then got around to start the HttpServer somehow:

[image]

I got my density messed up, cause I wanted beefier buttons, but `scrcpy` helped out.
After starting the HttpServer from the app, I can see it with `dumpsys` and browse this sweet api 🍾

Thanks!

One more thing though: could you explain what the ScreenSaver does?
I was looking for an option to:
1. TurnOFF the screen completely while not in use (TurnON on touch)
1.1 Like lock or show completely black image instead of ScreenSaver?
2. Start ScreenSaver after like 15s, spin the digital clock around
2.1 This is possible from settings, but there was an adb command, I cannot remember now

closed 2025-02-28 by Andoramb

RapierXbox, 2025-03-01:
The screen saver turns off the screen backlight after 45 seconds of no touch input. This means the screen uses way less power and becomes completely dark. I’ll probably add a setting for the delay in a future update. The clock is a good idea—I’ll likely include it in the same update.