RapierXbox/ShellyElevate #10: Can't connect to ADB

radclif, 2025-03-01:
I have enabled ADB according to the wiki. But I can not connect.

adb connect 192.168.23.80
failed to connect to '192.168.23.80:5555': Connection refused

I am running:
adb --version
Android Debug Bridge version 1.0.41
Version 35.0.2-12147458
Installed as /opt/homebrew/bin/adb
Running on Darwin 24.3.0 (arm64)

And Shelly Wall Display FW version 2.3.4 stable with is the latest firmware as of 2025-03-01.

pytekmobile, 2025-03-02:
same problem here.
I've updated the wall display to the lastest firmware and used the jailbreak info to enable the developer settings.
can't connect from my Mac mini or my MacBook Pro. Tried it in two different wifi networks.
Also tried a port scanner and it doesn't find an open TCP Port at 5555.
Did a hard reset and enabled it again. has no effect.
So I have opened the shelly and connected a usb a to usb c cable to it. with this it is working.
But it would be nice to connect to the shelly over wifi.

radclif, 2025-03-02:
I will do it the USB way tonight. Thanks!

RapierXbox, 2025-03-02:
For me, the developer settings worked. Did you make sure that you are on the same subnet and that Wi-Fi is enabled and connected? There is also the possibility to use the MTKClient version or the UART version you can find [here](https://github.com/RapierXbox/ShellyElevate/wiki/Jailbreak-using-UART--(Deprecated-dont-use!)).

RapierXbox, 2025-03-02:
Oh yeah, sometimes you need to reboot the device fully for the ADB changes to take effect.

RapierXbox, 2025-03-02:
Maybe your port is different. Could you go into the developer settings, navigate to system information, then scroll down to `persist.adb.tcp.port` and try connecting using this port?

pytekmobile, 2025-03-02:
thanks for the tipp, the system information didn't had a persist.adb.tcp.port or service.adb.tcp.port property.
after executing 
setprop persist.adb.tcp.port 5555
setprop service.adb.tcp.port 5555
in adb over the usb connection and a reboot it now works via wifi too.
Thx!

radclif, 2025-03-03:
@pytekmobile How did you get the USB connection to work? I have followed the guide at:
https://community.home-assistant.io/t/installing-3rd-party-apps-firmware-on-shelly-wall-display/657440

1. I disconnected the power from the Shelly wall display.
2. Diconnected the usb-c cable between the wall display and my macbook pro
3. ran the command "python mtk.py rl dump" in terminal
4. connected the usb-c cable between the wall display and my macbook pro
5. connected the power to the Shelly wall display.

Output:
Preloader - Status: Waiting for PreLoader VCOM, please reconnect mobile to brom mode

Port - Hint:

Power off the phone before connecting.
For brom mode, press and hold vol up, vol dwn, or all hw buttons and connect usb.
For preloader mode, don't press any hw button and connect usb.
If it is already connected and on, hold power for 10 seconds to reset.

...........

Port - Hint:

Power off the phone before connecting.
For brom mode, press and hold vol up, vol dwn, or all hw buttons and connect usb.
For preloader mode, don't press any hw button and connect usb.
If it is already connected and on, hold power for 10 seconds to reset.

..........Preloader
Preloader - [LIB]: Status: Handshake failed, retrying...

Port - Hint:

Power off the phone before connecting.
For brom mode, press and hold vol up, vol dwn, or all hw buttons and connect usb.
For preloader mode, don't press any hw button and connect usb.
If it is already connected and on, hold power for 10 seconds to reset.

...........

Port - Hint:

Power off the phone before connecting.
For brom mode, press and hold vol up, vol dwn, or all hw buttons and connect usb.
For preloader mode, don't press any hw button and connect usb.
If it is already connected and on, hold power for 10 seconds to reset.

...........

I have tried several times. Also did a factory reset of the Shelly WD. Tried 3 different USB-C cables. I'm powering the Shelly WD with 230v.

pytekmobile, 2025-03-03:
@radclif I haven't used the python script in the link.
I've dissembled the wall display. connected the usb c Cable to the display. powered on the wall display and waited until it was fully booted and then connected it to the Mac.  In this way I was able to communicate through adb with the device and could set the port properties for adb tcp.
Two things are very important:
# 1 connect the cable to the Mac after the wall display has fully booted
# 2 don't use a usb c -> usb c cable with your Mac. it doesn't work. I had to use a usb A to usb C cable with an usb switch. this sounds crazy, but that's the only way it works.

It was also very helpful to enable the developer option "Enable default launcher and SystemUI". With this setting you get the android status and navigation bar and you immediately see if the usb cable is connected with a small icon in the status bar on the top left.

radclif, 2025-03-03:
@pytekmobile A thousand thanks for your help!

Changing the cable to a USB-A to USB-C cable and using a Satechi dongle with my Macbook solved it.

I have now installed ultra-small-launcher.apk and ShellyElevateV2.apk sucessfully and also set the adb -d shell setprop persist.adb.tcp.port 5555 & adb -d shell setprop service.adb.tcp.port 5555 so I can access the Shelly WD with adb connect [IP].

I downloaded https://www.fully-kiosk.com/en/#download-box to point it to a clean WEB UI made to look good in the wall display I'm coding that is using the Home Assistant API to turn on/off lights, unlocking the front door for 5 seconds and to display a keypad for arming/disarming my Home Assistant Alarmo system. The Shelly is mounted next to my front door instead of the original light switch.

I went this route because the Home Assistant integration in the Shelly WD was waaaaay too slow. Taking about 10 seconds to update the UI when a button was pressed. It might be because I have configured my HA to be very heavy when it comes to the UI. The standard UI is hideous regardless of theme so I'm using Lovlace-minimalist-UI and a lot of other stuff to make it look like it's not designed by an engineer. 😆 [Screenshot of my Home assistant](https://imgur.com/a/sWKsJV9)

radclif, 2025-03-05:
[Result of the project](https://imgur.com/a/I2wa35t) Thanks a lot!

RapierXbox, 2025-03-06:
Ok, this issue clarified some things. It looks like enabling it just enables it for USB. I am going to update the guide and look for another way to hack it without disassembling. I am also going to reopen this issue until I have updated the guide. I’m not really sure what your setup is right now. Are you running ShellyElevate in the background for the relay functions and so on, or are you fully using Fully Kiosk?

radclif, 2025-03-07:
@RapierXbox I ended up just using the Fully Kiosk for now. I will take a look at ShellyElevate later, for now I put another Shelly 1PM mini behind the Shelly WD to control the "dumb" spotlights in that room.
I also used "adb shell wm overscan 0,0,0,-202" to get rid of the Android navigation bar at the bottom for anyone curious.

RapierXbox, 2025-03-07:
How did you get a shell on the device. Using adb over usb or is there a feature in mtk_client?

pytekmobile, 2025-03-08:
With adb over usb and the command  „./adb shell“ you get Shell Access on the Wall Display.
I‘ve Build my own android App on the base of ShellzElevate with openHAB Connection. Here you can see the result https://imgur.com/a/dM31rYC
My next step is to try building the app with compose mutliplatform to get it run on iOS devices too.

RapierXbox, 2025-03-09:
I fixed the guide to use USB over ADB. Sadly, I didn’t find any way to enable ADB over TCP from the developer settings. I will still be searching for ways to hack it fully without disassembling.

closed 2025-03-09 by RapierXbox

radclif, 2025-03-09:
@pytekmobile Very nice app you've made! Just as a sidenote, adb shell works over wifi as well for me.

radclif, 2025-03-25:
> Hi, I also wanted to install fully kiosk browser, which version are you using? Only crashes occur from time to time

https://www.fully-kiosk.com/files/2025/02/Fully-Kiosk-Browser-v1.58.apk

artur90226, 2025-03-25:
> > Hi, I also wanted to install fully kiosk browser, which version are you using? Only crashes occur from time to time
> 
> https://www.fully-kiosk.com/files/2025/02/Fully-Kiosk-Browser-v1.58.apk

Android Webview Didn't you update which version? I think could have something to do with it

RapierXbox, 2025-03-25:
Could you let us know your WebView version and package name? You can check it by first determining your WebView package name with:  

```sh
adb shell pm list packages | grep webview
```

Then, use the package name to get the version with:  

```sh
adb shell dumpsys package com.google.android.webview | grep versionName
```
<sub>make sure to replace the package name with yours<sub/>

Thanks!

artur90226, 2025-03-26:
> Could you let us know your WebView version and package name? You can check it by first determining your WebView package name with:
> 
> adb shell pm list packages | grep webview
> Then, use the package name to get the version with:
> 
> adb shell dumpsys package com.google.android.webview | grep versionName
> make sure to replace the package name with yours
> 
> Thanks!

Probably you don't mean me? 
Now for a few days without problems, no crash

artur90226, 2025-03-28:
I noticed today that webview also crashes

paolo-ermit, 2025-04-01:
Just a note: I could now connect with a usbC-usbC cable and I had to use a usbA-usbC cable.
I think that it worth mentioning in the guide.

artur90226, 2025-04-25:
Something new about android webview crash?