RapierXbox/ShellyElevate #11: Setting timeout for screensaver

heartkingz, 2025-03-14:
Setting the timeout for the screensaver to kick in does not work, using the latest release. 
It just starts the clock after a second or two, even though I set the timeout to a higher number. 
When using the screensaver type "turn off screen" doesn't work either. 
It always starts the clock.

RapierXbox, 2025-03-14:
Hi Heartkings,

I am aware of this issue. It happens because I accidentally saved the screensaver ID and the delay to the same preference. I am really sorry. This will be fixed in the next release. It will probably come out tomorrow or Sunday. because I also want to fix another bug where the HTTP server sometimes doesn’t respond. I will try my best to finish the update today!

heartkingz, 2025-03-14:
Don’t feel sorry and take your time! I really appreciate your work and looking forward to the next release.

closed 2025-03-15 by RapierXbox

heartkingz, 2025-03-15:
Thank you so much for the update! works much better, the timeout works and the screen turns off when idle. 
The only downside I have noticed is that when the screen is dark and you tap it, it registers the tap directly within the home assistant page that's "behind" it, which makes me turn on or off a light for example while I just want to wake it up. 
It would be better if you need to tap the screen first to wake it up without it registers touch input. The  original Stargate app does that too.

RapierXbox, 2025-03-15:
Haven't noticed that yet, but I will fix it in the next update!