RapierXbox/ShellyElevate #12: MQTT support & more HTTP endpoints

dbochicchio, 2025-03-14:
Hi,
I just installed it on my X2, to replace an old Fire Tablet running Fully Kiosk. Great job, I love this solution, since the X2 screen is incredibly crisp.
I don't run Home Assistant, but my own UI.

The only thing I'm missing are a couple of commands/events:
 - get/set screen status (my use cases: turn the screen on and invoke a function via web sockets when the doorbell is pressed to show the cams / turn it off when house is in away, night or vacation mode)
 - get the screen status inside Javascript (as in Fully Kiosk), to pause updates when the screen/app is off and resume when on again.
 - the ability to refresh the current page, to resume from errors from time to time

MQTT support will be really appreciated as well. Events instead of polling is always preferred. Thanks!

RapierXbox, 2025-03-14:
Hi dbochicchio,

Thanks for your feature request! I really appreciate it, as it gives me something to work on. I will take a look at the three points and try to implement them at least by V2.2.2. I'm also looking forward to MQTT, but for now, my focus is on enabling full app installation via the touch interface.

Could you also tell me how you jailbroke the device and maybe send some pictures of the insides?

Thanks!

dbochicchio, 2025-03-15:
I used the F/H sequence, I got the ADB working with USB and that's all. No need to open it.

RapierXbox, 2025-03-15:
Does it have a external usb?

dbochicchio, 2025-03-15:
Yes, external usb on the right. I could update the doc and add a picture, if you think it'd be useful.

dbochicchio, 2025-03-16:
I got some time today to thinker with it again. So, a couple of new feature requests:
 - get/set screen brightness (I specifically opened a bug for this)
 - get/set current displayed URL
 - kill app
 - reboot device
 
Thanks!

closed 2025-03-18 by RapierXbox