RapierXbox/ShellyElevate #14: Access to proximity sensor status

dbochicchio, 2025-03-16:
According to spec, the device has a proximity sensor (and a G-sensor).

Access to the status will be appreciated. Also, an option to turn off the screensaver when the proximity sensor detects a movement could be useful, to glance at the screen without touching it.

heartkingz, 2025-03-16:
Really? Pretty strange the original firmware doesn't make use of that proximity sensor then!

RapierXbox, 2025-03-16:
The original display doesn’t have one. Would you be willing to test some beta firmware for it? Sadly, I don’t have an X2. Writing the code should be relativly easy.

dbochicchio, 2025-03-16:
Sure thing.

simllll, 2025-08-01:
Any news on this? I'm still on the original firmware, but the display currently has to get a touch to show something. My one and only feature request that is misisng right now is exactly this functionality ;-) If i come closer to the screen, it should wake up automatically (like the sonoff display does for example).

dbochicchio, 2025-08-01:
I’m working on a fork and I’ve added this feature. I think I’ll send a PR in a couple of days.

Unfortunately the proximity sensor is not very precise, but it works.

RapierXbox, 2025-08-14:
This has been added in the latest version, though I can’t fully confirm it works yet. If you run into any issues, feel free to reopen this issue.