RapierXbox/ShellyElevate #15: Black Screen

CKL1976, 2025-03-17:
I did managed to load everything onto my Wall Display and it has been working like a charm. However now the screen is dead, the device is booting and show the Shelly logo, but nothing happens after that.

The device is still reachable via ADB

Any suggestions as to what has happened and how to solve it ?

RapierXbox, 2025-03-17:
It's common for the autolaunch to not work sometimes. Did you make sure to install the app with the -g flag? What happens when you start the app manually?

CKL1976, 2025-03-17:
I did install with -g flag and also tried with -r flag.. Installs without any problems. Manual start of the app works.

But with the recent version I also now see problems with the display freezing after pushing 2-3 buttons in my dashboard. There are no animations running or anything to slow down the dashboard...

RapierXbox, 2025-03-17:
Also, for the autostart, are you sure Lite mode isn't enabled?

CKL1976, 2025-03-18:
I disabled Lite mode and now it starts....

It seems like with Lite mode enabled the device is less "responsive". I will keep testing :-)

RapierXbox, 2025-03-18:
Hmm… Lite mode only disables the autostart of the GUI. Maybe there is some glitch causing the application to launch twice when you start it normally. Try to find steps to reproduce the error and also check Logcat when it happens.

heartkingz, 2025-03-19:
With lite mode disabled and automatic brightness turned off in android settings and enabled via the shellyelevate app the screen does respond normally. I don’t know if this has something to do with the latest update but it seems like the screen does respond better with lite mode turned off.

androidvoorlars, 2026-02-17:
I have installed v2 sucessfully on to Shelky wall display (small) and all is working well. A few hours later I decided to do the same with the second display I have but unfortunately that didn't turn put well. All goes well, it discovers HA Ip but once saved it hangs on the screen with the HA logo. So before the screen I usually get to enter my credentials. Tried to re install via adb but no luck. Any ideas?