RapierXbox/ShellyElevate #16: Power supply

Sylensky, 2025-03-19:
Hi,

Kudos for reverse engineering this. I am thinking on getting a wall display but i do not have a wall socket available and wanted to use the inbuilt usb type-c connector for it.

Personally i checked the connector from the wall socket  and iam able to supply 5V via the connector pins directly. Suprisingly the display does not draw alot of current while running. Only booting up occassionally can spike to 1A.
Apparently i did have limited time to disassemble one display from a colleague and was not able to test the usb commector.

Did someone try to power the display using the inbuilt usb connector?

RapierXbox, 2025-03-20:
Powering the standard wall display from the built-in USB is challenging for a couple of reasons. First, accessing the USB port requires disassembling the display, and second, the display doesn't boot from USB. This behavior is reaaaaly weird, at least to me. When connected to USB, it doesn't boot, even though I can measure 5V on the daughterboard pins. It only starts booting when the power is initially supplied from the daughterboard and then switches to the USB-C port. This is also why you need an external power supply when installing the ShellyElevate app. The larger display might boot, but I haven't had the opportunity to test it.

Sylensky, 2025-03-21:
Thanks for clarifying. My choice tends towards the larger X2 display due to multiple reasons.
While powering via USB being the largest of it.

closed 2025-03-21 by Sylensky

dbochicchio, 2025-03-21:
X2 is ok with just USB. I have it powered up on the USB port of my PC, to test it.