RapierXbox/ShellyElevate #21: X2 autostart does not work

kallejoken, 2025-05-03:
i have successfully rooted the x2, everything works - but unfortunately the autostart does not, is it possible to implement this?

RapierXbox, 2025-05-09:
Could you try rebooting the device, and if Shelly Elevate doesn’t start, run the following command:
`adb shell am broadcast -a android.intent.action.BOOT_COMPLETED`

You can run logcat for just Shelly Elevate using:
`adb logcat --pid=$(adb shell pidof me.rapierxbox.shellyelevatev2)`
or for the whole system using:
`adb logcat`

Nick-Sills, 2025-05-09:
Not sure if this is relevant

<code>adb shell am broadcast -a android.intent.action.BOOT_COMPLETED
Broadcasting: Intent { act=android.intent.action.BOOT_COMPLETED flg=0x400000 }
Exception: java.lang.SecurityException: Permission Denial: not allowed to send broadcast android.intent.action.BOOT_COMPLETED from pid=1675, uid=2000</code>

Nick-Sills, 2025-05-09:
`adb logcat` works, but given the response `android.intent.action.BOOT_COMPLETED` I'm not sure if the output from that helps?

RapierXbox, 2025-05-09:
Oh sorry, run `adb root` before to get root privileges!

Nick-Sills, 2025-05-09:
Better now:

```adb shell am broadcast -a android.intent.action.BOOT_COMPLETED
Broadcasting: Intent { act=android.intent.action.BOOT_COMPLETED flg=0x400000 }
Broadcast completed: result=0

But ShellyElevate still  isn't running so I'm not sure how the logs are helpful ? 
Is running logcat useful at this time, or is there have I missed something?

RapierXbox, 2025-05-09:
The Application should definitely start at this point. Logcat could be useful. Run it for the whole system and try to filter out most of the logs so they aren't so long. Maby it crashes early on?

Nick-Sills, 2025-05-10:
@RapierXbox - `send-broadcast denyed` 

```bash
05-10 08:02:25.144   635   794 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.BOOT_COMPLETED flg=0x400010 } to plugin.sprd.contactsdefaultcontact/.PreDefineContactsReceiver
05-10 08:02:25.144   635   794 D BroadcastQueue: Skipping delivery of ordered [background] BroadcastRecord{9c8f563 u0 android.intent.action.BOOT_COMPLETED} for whatever reason
05-10 08:02:25.145   635   650 D PowerHintScene: #### targetApp: me.rapierxbox.shellyelevatev2 , callerApp: null, callerUid:0
05-10 08:02:25.145   635   650 D PowerController.BgClean: in autolaunch black list: me.rapierxbox.shellyelevatev2, callingPackage = null, reason = send-broadcast denyed!
05-10 08:02:25.145   635   650 D BroadcastQueue: Skipping delivery of ordered [background] BroadcastRecord{9c8f563 u0 android.intent.action.BOOT_COMPLETED} for whatever reason
05-10 08:02:25.146   635   650 D BroadcastQueue: Finished with ordered broadcast BroadcastRecord{9c8f563 u0 android.intent.action.BOOT_COMPLETED} 
```

Nick-Sills, 2025-05-10:
@RapierXbox Fixed it  😄 

From the Android home screen: Settings -> Battery  -> Auto Launch Management -> Enable ShellyElevate2

[image]
[image]